htotoo/ESP32-Portapack #16: After burning, there is no WIFI signal. What is the reason? Do I need PP to transmit WIFI

WXLN, 2024-09-12:
[image]

htotoo, 2024-09-13:
You may have to wait for a bit before the AP shows up.  (max 2 min in the worst case)

You should check the logs. Plug it in computer (via the usb serial interface, not the otg one), and open the serial port it creates (com12 in your case). Then press reset, and copy here what it writes.

htotoo, 2024-09-13:
Also try the new 0.9,3 edition, not the 0.8

WXLN, 2024-09-15:
Later, I replaced it with an 8MB flash memory module, which was able to use the network normally, but the 4MB module still couldn't function properly
[image]

htotoo, 2024-09-15:
Yeah, the flash size designed to a minimum 16mb flash. So it is wonder it works with 8mb!
Don't try to update with ota,.it may break it.

closed 2024-09-22 by htotoo