htotoo/ESP32-Portapack #34: Cannot download files from SD Card

Damien0505, 2024-12-02:
I have connected an ESP32-S3 flashed with the latest firmware using the web installer.

The installation and connection is successful, I can see GPS data in the web interface and control the H4M.

The problem I have is when trying to download a file from the H4M, I receive the following:

  File read error
  Command executed
  < filesize /FREQMAN/POCSAG.TXT

Upload works fine, and if I remove the SD card from the H4M and read it on a PC it is fine too.
Would try file download under v0.13 but "Open file manager" appears greyed out.

HRF/PP:  H4M
H4M firmware:  n_241201
ESP32PP firmware:  0.14

htotoo, 2024-12-02:
You connected it via USB or I2C?

What is the error you get there?

htotoo, 2024-12-02:
Yeah, sadly if the i2c is faster than the wifi, it happens. Good news, this is at least indicated :)
This is in my todo list, to choose a slower rx when it is over i2c.

Damien0505, 2024-12-02:
Thanks for the speedy reply.  Happy for you to either close this issue, or mark it suitably.

htotoo, 2024-12-02:
Keep it open, till there is no solution

closed 2024-12-28 by htotoo